sinara-hw/Phaser #121: replace TRN 3-1222

hartytp, 2020-09-24:
phaser burns off a lot of power. We should be careful not to be wasteful and let the usage creep up unneccecarily.

The TRN 3-1222 uses 480mW (40mA) quiescent, which is a lot bearing in mind that we only need ~20mA from each rail.

Replacing this with the 1W converter in the same family would halve the quiescent current, which would help a lot.

gkasprow, 2020-09-24:
True. We don't need that much.

gkasprow, 2021-03-11:
I will replace them with dedicated buck and inverting synchronous converters

gkasprow, 2021-03-11:
Not that easy. the negative sync converters work up to -5.5V.
And we need to create 12V out of input voltage which can be less than 12V.
We have to use SEPIC converter, for example LT8582 which delivers +/- 12V. We can also go for LTM8049.

gkasprow, 2021-03-11:
But it is overkill.
Let's go for [NMJ1212SC](https://datasheet.octopart.com/NMJ1212SC-Murata-Power-Solutions-datasheet-6387223.pdf). it consumes 250mW of quiescent power.

closed 2021-03-11 by gkasprow

gkasprow, 2021-03-11:
we can also go for [MTU1D1212MC](https://www.murata.com/products/productdata/8807030456350/kdc-mtu1.pdf?1612841424000), it has higher efficiency and lower ripples, but has a non-standard footprint.
[NMV1212SC](https://datasheet.octopart.com/NMV1212SC-Murata-Power-Solutions-datasheet-6396052.pdf) has standard pinout and higher efficiency. The same with TRN 1-1222 which has better output stability.

gkasprow, 2021-03-11:
Let's use TRN 1-1222 then