sinara-hw/Phaser #123: 1:2 Balun does not double the voltage

Jacky2k, 2021-02-03:
There is a comment in the schematic (PDF page 10) about the TR5A balun, saying:
`Balun will double that voltage` 
Please correct me if I am wrong, but that is not correct if the termination on the secondary side is 50R.
The 1:2 ratio relates to the impedance, not the voltage. As the power is (roughly) constant, the voltage will increase by sqrt(2), while the current will decrease by sqrt(2), doubling the impedance. But the voltage is only amplified by sqrt(2).

gkasprow, 2021-03-11:
@hartytp shall we use other balun?

hartytp, 2021-03-12:
Not unless there is a specific reason to change it (and I'm not aware of anyone raising one). I believe the post above was a comment that the annotation is incorrect rather than a request to change the balun

Jacky2k, 2021-03-12:
I just noticed that the annotation is wrong. I don't know if a 1:2 voltage transformation was intended.

gkasprow, 2021-03-13:
ok, right.

closed 2021-03-13 by gkasprow