sinara-hw/Phaser #125: Phase noise measurement

sotirova, 2021-03-01:
# Setup

- EEM0 of a Kasli 2.0 connected to EEM0 of a phaser baseband board
- Kasli 125 MHz ext ref from CLK SYN of R&S SMA 100A
- Phaser 125 MHz ext ref from RF output of R&S SMA 100A; output level 0 dB
- Generate a single tone from RF0 of phaser at 150 MHz, output level -9.1 dBm
- Connect RF0 of phaser to R&S phase noise analyser

# Measurement results

First look at phase noise when phaser is using its internal clock vs when it is given an external 125 MHz clock as described above. For both of these measurements Kasli was connected to an external 125 MHz reference as above.
[image]

For reference, here is also the phase noise from the Kasli MMCX clock outputs when using the Kasli internal clock vs when Kasli takes an external 125 MHz reference.
[image]

hartytp, 2021-03-02:
> Kasli 125 MHz ext ref from CLK SYN of R&S SMA 100A

Interesting. Do you know/did you ever measure what the phase noise spectrum of that output is? I recall it being significantly worse than the main output, but can't find any info in the spec sheet.

hartytp, 2021-03-02:
Anyway, that's lovely data and generally better than I'd expected. From a physics perspective, that's plenty good enough for everything I want to do, which is great.

In terms of background curiosity/calibrating expectations for what should be doable for this kind of design, the AD9910 noise graphs with and w/o PLL are copied below. My understanding is that DDS chips like the AD9910 are generally about as good as one gets for this kind of digital synthesis so are a pretty good reference.

[image]

Looks like phaser is within 10dB of the AD9910 at all frequencies, a bit better than the 9910 at higher frequencies and a bit worse at lower frequencies, but broadly comparable throught.

[image]

Phase with PLL is something like 20dB worse than the AD9910 w/o PLL for close-in noise with less than 10dB between them for broader-band noise. 

As a general point of interest, I'd be curious to know what the noise spectrum of phaser looks like without the PLL, but that requires a bit of reconfiguration is is not something we need to worry about.

hartytp, 2021-03-02:
Anyway, good work @sotirova! I don't see anything suspicious here (the DAC data sheet doesn't have phase noise plots so doesn't give us anything to directly compare with, but I don't see anything in the above to suggest we're doing worse than I'd expect).

closed 2021-03-02 by hartytp

sotirova, 2021-03-02:
Thanks for the comments @hartytp!

> Interesting. Do you know/did you ever measure what the phase noise spectrum of that output is? I recall it being significantly worse than the main output, but can't find any info in the spec sheet.

Here's the phase noise of the synth RF OUT and CLK SYN for reference - indeed the CLK SYN looks worse than the main RF output
[image]

hartytp, 2021-03-02:
> Here's the phase noise of the synth RF OUT and CLK SYN for reference - indeed the CLK SYN looks worse than the main RF output

What carrier frequency?

sotirova, 2021-03-02:
> What carrier frequency?
125 MHz both, as used in the setup

hartytp, 2021-03-02:
anyway, good to know that the CLK SYN phase noise is not a limitation here (although not by that far)

pathfinder49, 2021-04-14:
I've performed similar measurements for the Phaser-Upconverter variant.

My setup:
* 10 MHz and 125 MHz reference frequencies generated by R&S SMA 100 A
* R&S FSWP phase noise analyser (using 10 MHz external reference)
* Kasli v1.1 (using 125 MHz external reference)
* Phaser upconverter (this uses the kasli clock distribution or dedicated 125 MHz external reference, see below)
* For measuring LO:
    * All amplitudes and frequencies set to zero
* For measuring RF output
    * DAC-NCO at 200 MHz
    * DUC at 130.3 MHz
    * Kasli oscillator at 7 MHz

|   \   |   Using external reference distributed by Kasli   |   Using dedicated phaser reference   |
|   ---   |   ---   |   ---   |
|   Phaser Reference Clock    |   [image]   |   [image]   |
|   Phaser 2875 MHz Local Oscillator   |   [image]   |   [image]   |
|   Phaser 3.2123 GHz output   |   [image]   |    [image]   |

At ~3 GHz the output phase noise is dominated by the TRF-LO phase noise. The phase noise can be improved substantially by using a dedicated frequency reference.