sinara-hw/Phaser #129: Phaser Upconverter - TRF lock failure for high f_PFD

pathfinder49, 2021-03-26:
I'm testing a Phaser v1.0 with a Kasli v1.1 and am observing intermittent TRF lock failures when using f_PFD above 20 MHz.
The exact threshold for intermittent trf locks varies between trf0 and trf1.

- trf0: intermittent lock with f_PFD > 125 / 6 MHz = 20.83 MHz  (tested smaller integer divisions of 125 MHz)
- trf1: intermittent lock with f_PFD > 125 / 4 MHz =  31.25 MHz (tested smaller integer divisions of 125 MHz)

Is this expected behaviour of phaser?
What f_PFD was phaser designed around? 

Looking through the data-sheet there are examples up-to 40 MHz. However, I haven't found an explicit maximum frequency. 

Comparing the phaser loop-filter to the data-sheet recommendation for different f_PFD frequencies doesn't help much as they are dissimilar.
[image]
[image]

pathfinder49, 2021-03-26:
Aside, high F_PFD clocks are desirable to reduce the LO phase noise.

hartytp, 2021-03-26:
> Looking through the data-sheet there are examples up-to 40 MHz. However, I haven't found an explicit maximum frequency.

Isn't that 
[image]

pathfinder49, 2021-03-26:
I've patched the core-device driver to only enable the RF and LO outputs after the trf lock has been acquired (as recommended by the data-sheet). With this modification, both TRFs can operate upto 31.25 MHz, but not higher integer divisors of 125 MHz. This is still below the data-sheet expectation. However it removes the channel discrepancies.

hartytp, 2021-03-26:
> What f_PFD was phaser designed around?

The design was all done via GH issues, so the details are there. See the discussion in https://github.com/sinara-hw/Phaser/issues/49

hartytp, 2021-03-26:
If you go on the upconverter page at TI there is a link to the loop filter simulation tool. It's pretty easy to install and use to verify a configuration.

hartytp, 2021-03-26:
What VCO frequency, Icp etc are you using?

pathfinder49, 2021-03-26:
Thanks for the link :) From the discussion the intended operating f_PFD is 62.5 MHz.

Looks like no-one got around to the phase noise testing https://github.com/sinara-hw/Phaser/issues/49#issuecomment-555228365

Aside: the current artiq-driver uses f_PFD = 6 MHz

hartytp, 2021-03-26:
The VCO slop calculation looks a bit funny to me. If I understand correctly the filter design tool implies this should be 45MHz/V 
[image]

...but playing around with the loop, that shouldn't induce instability afacit.

pathfinder49, 2021-03-26:
> What VCO frequency, Icp etc are you using?

I'm using f_VCO = 2.875 GHz. So far I've only looked at the clock configuration (see https://github.com/m-labs/artiq/issues/1643). Until now I've left Icp, etc. at the driver default (the implementation may be similarly out-of-spec to the clocks though).

pathfinder49, 2021-03-29:
I've now got f_PFD locking reliably at 62.5 MHz. The previous limitation was due to the PFD logic clock, which was running too fast.

Closing in favour of the existing phase noise measurement issues.

closed 2021-03-29 by pathfinder49