sinara-hw/Phaser #130: Upconverter Output Power

pathfinder49, 2021-04-09:
From the TRF data-sheet and the Phaser schematics it looks to me like the MMCX LO output should be at +3 dBm. However, I only get -27 dBm. Is this expected?

Data-sheet and phaser schematics:
[image]
[image]

My setup:
* Phaser v1.0
* I'm using TRF1. This has LO OUT P and LO OUT N available via MMCX connectors.
    * LO OUT N terminated with 50 Ohm (doesn't make a significant difference)
    * LO OUT P connected to a 50 Ohm scope/spectrum analyser (tried both)
* LO frequency configured as described in the data-sheet snippet above
* Using frequency reference provided by Kasli

jordens, 2021-04-09:
Divider or buffer disabled?

[image]

jordens, 2021-04-09:
How does it change with output bias?

[image]

pathfinder49, 2021-04-09:
@jordens Thank you for the suggestions :) This is with both PWD_OUT_BUFF and PWD_LO_DIV written to 0 (enabled). Changing the output bias also changes the output by ~1 dB.

Curiously, setting `PWD_LO_DIV = 0` and `PWD_OUT_BUFF = 1` gives a 7 dB power increase (to -20 dBm). However, this is still well below the nominal +3 dBm.

jordens, 2021-04-09:
VCC_LO1 and VCC_LO2 are high with 5V > 3.3V. Probably died.

jordens, 2021-04-09:
Also potential reason for the high consumption.

pathfinder49, 2021-04-09:
> VCC_LO1 and VCC_LO2 are high with 5V > 3.3V. Probably died.

Good spot! CC @gkasprow 
[image]
I guess I can write off my other up-converter testing until this is fixed. Surprisingly The chip still seems to mostly work. The mains quirks I've noticed so far is the LO output power (see above) and similarly reduced RF output power.

jordens, 2021-04-09:
If the LO output stage didn't die yet, it's certainly biased low. Try biasing with 5V (and maybe higher resistance).

This reaffirms the old lessons: (a) better and thorough review (b) characterize and understand fully before calling probably-good-enough. No surprises here.

hartytp, 2021-04-09:
> VCC_LO1 and VCC_LO2 are high with 5V > 3.3V. Probably died.

AFAICT that's above the recommended voltage but still within abs max, so it's unlikely that this has resulted in damage

[image]

hartytp, 2021-04-09:
[image]

hartytp, 2021-04-09:
anyway, needs fixing for a future release (easy air wire patch) but not clear if this will actually cause damage.

gkasprow, 2021-04-09:
the VCC_LOx are supplied via 1k Ferrite beads which have 1.25Ohm resistance. So the voltage is even lower than 5V.

hartytp, 2021-04-09:
Yeah, but only by 150mV. But, I agree, it does seem like this is within abs max ratings, so not ideal but unlikely to cause damage...

[image]

jordens, 2021-04-09:
I'd be very careful with that interpretation of the datasheet. Might be a typical datasheet inaccuracy. For the TRF3722 (in some ways the successor) they do differentiate and have abs max 3.6 V for those supplies. Also for the TRF372017 they only list one 3.3V supply as being 5V-tolerant. Presumably the others aren't that tolerant.
I vaguely to remember that I saw surprisingly high output powers in the beginning which then permanently dropped after some time.
Go ahead and give it a try.

pathfinder49, 2021-04-09:
I'll try patching this on Monday and see if the output power improves. There is no convenient P3V3A plane on the top layer. Therefore, the supply will need to share a via with another TRF pin. At 1 GHz, the via will have a ~1 Ohm impedance, which is significantly less than the 1.6 kOhm of the ferrite beads. The plan would be:
1. Cut the connection between the VCC_LOx bead and the P5V0A supply
2. Add an air wire from the bead to a P3V3A via.
The most practical via choices seem to be: 
* VCC_LO1 from VCC_DIG via (not ideal but seems better than VCC_PLL)
* VCC_LO2 from VCC_VCO2 via

Power wise, the budget already accounts for the TRF372017 in the patched state.

jordens, 2021-04-09:
Just rotate the beads and wire up their inputs.

pathfinder49, 2021-04-12:
I've patched ch1 on my board as described above. 

The patch has greatly improved the LO output power. I now get -1.7 dBm. This is in-line with the data-sheet expectations.

Regarding the RF output: The maximum output power I can get is roughly -12 dBm. This is irrespective of applying the patch. I would expect a peak output power of at least -5 dBm (see below). The RF output seems to be missing ~7 dB.
 
Configuration:
* 1 Kasli oscillator set to 0.99 asf and 7 MHz (other oscillators disabled)
* DUC set to 143 MHz (for 150 MHz total)
* Mixer disabled
* LO set to 2875 MHz

From the DAC data-sheet I would expect ~0 dBm DAC output. 
[image]

The TRF would attenuate this by <4 dB
[image]

pathfinder49, 2021-04-12:
For others needing to patch this, here is a picture. As the components are small (0402?), I found it easiest to cut the P5V5A traces. 
[image]

jordens, 2021-04-12:
Check how much loss you expect from #119 vs what the datasheet assumes. And also `bufout_bias` and other settings like the DAC full scale output current.

Jacky2k, 2021-04-13:
I can confirm that the LO output power is within spec after applying the fix. I rotated the beads instead of cutting the wires on the PCB, which is less invasive and was fairly easy with a microscope and a hot-air-station, image below.
However, the problem with the low output power still persists. I am willing to help and invest some time in the problem, as I would appreciate a higher output power of the Phaser. Any tips are welcome.
[image]

pathfinder49, 2021-04-13:
I've streamlined my setup to reduce cable losses. Using only a CBL-0.5M-SMNM+ cable from the Phaser output to the scope, increases the RF power with the settings above to -10.2 dBm. I've also probed the signal chain using the Keysight N2795A 1GHz, 1pF probe. 

* The differential pairs coming from the DAC are at - 5.8 dBm relative to ground. As their voltages add, the signal in the pair is therefore 0.2 dBm. This is Consistent with the expected power from the DAC.
* After the reconstruction filter I get -11.7 dBm and –11.9 dBm for the two halves of the differential pair. This corresponds to roughly -5.7 dBm for the pair.
* The Typical TRF attenuation st this frequency is 2.7 dB. This brings the signal to -8.4 dBm
* The typical attenuator insertion loss at 3 GHz is 1.8 dB, bringing the expected signal to -10.2 dBm.

In conclusion, the LO output power was due to the supply voltages. The RF output power at 3 GHz is limited by hardware to -10 dBm. The majority of the losses (6 dB) occur in the reconstruction filter.

Jacky2k, 2021-04-13:
It seems like the TRF is working as expected. I set everything up as @pathfinder49 described. Except I don't know what you mean by turning off the mixer. How can you turn the mixer off? Is there a bypass?
Anyway, in this case I get about -10 dBm output power with 7 MHz Osciallator, 143 MHz DUC and 2875 MHz LO. I measured 108 mVrms at the I and Q inputs (differentially), which corresponds to -6.3 dBm. Subtracting the 2.7 dB loss in the mixer and the 1.9 dB loss in the attenuator, I get -10.9 dBm. There are of course some uncertainties, like additional losses in wires, cables or the fact that I was using two ordinary oscilloscope probes which are only speced up to 250 MHz.
In my calculation I am only missing ~5 dB from the DAC, which may be the reconstruction filter.

Jacky2k, 2021-04-13:
Ohh... I think that matches pretty well.
One other interesting thing is this:
[image]
This is a plot of the output power (dBm) vs. the DAC output frequency. The frequency is swept by changing the DUC frequency, everything else is the same as before. If the reconstruction filter is the cause for the additonal 5-6dB losses, shouldn't the amplitude go up for lower frequencies?

pathfinder49, 2021-04-13:
> How can you turn the mixer off? Is there a bypass?

The mixer is an integrated feature of the DAC. Unfortunatly, the corevice driver lacks much of the documentation for this.

> Ohh... I think that matches pretty well.
> One other interesting thing is this:
> [image]
> This is a plot of the RF output power (dBm) vs. the DAC output frequency (MHz). The frequency is swept by changing the DUC frequency, everything else is the same as before. If the reconstruction filter is the cause for the additonal 5-6dB losses, shouldn't the amplitude go up for lower frequencies?

This plot is not related to the hardware. What you are measuring here is the gain of the software DUC of Phaser.

hartytp, 2021-04-13:
Filter design is https://github.com/sinara-hw/Phaser/issues/4

Jacky2k, 2021-04-13:
> This plot is not related to the hardware. What you are measuring here is the gain of the software DUC of Phaser.

Can you explain this? I am measuring the RF output power, not the output of the DAC. Does the DUC include an amplitude compensation? Even if, I think the final output amplitude should be constant over frequency, right?

pathfinder49, 2021-04-13:
@Jacky2k Unfortunately, the artiq driver for Phaser (I assume that's what your'e using)  is not good at explaining these things. I've also had to make a fair number of changes/to the coredevice driver to make features such as the DAC-NCO usable. I'll consolidate this into a PR once I've completed my testing. The Phaser issues in the artiq repository may be helpful to you.

Jacky2k, 2021-04-13:
But the NCO should not touch the amplitude? Or what am I missing here?

pathfinder49, 2021-04-13:
@Jacky2k please open an issue on artiq. Lets discuss somewhere where others can find it :+1:

jordens, 2021-04-13:
Unfortunately that was very bad advice. Please don't open issues on artiq for these things. It's not a lab notebook, documentation hub, or user forum. Nor is there evidence that it's got to do with the duc. Nor is the duc part of artiq.

closed 2021-04-13 by pathfinder49

pathfinder49, 2021-11-15:
@hartytp The low output power from phaser forces us to chain multiple amplifiers to achieve the desired signal powers. This results in boradband noise from the pre-amplifiers being amplified to appreciable levels. In my case we can filter out most of this noise. However, it would be nice for future revisions to have larger output powers from phaser upconverter (say 0 to +10 dBm).

hartytp, 2021-11-15:
@pathfinder49 I'm a bit confused by your post...(a) because unless I'm missing it's only tangentially related to the above discussion (b) because there is a lot of context missing about your application/what you're trying to achieve here (c) because it's not clear to me what outcome you want -- is this a question? concrete request for a change to the design? statement about what you would like in an ideal world to inspire future discussion? (d) because I can't tell from the above what level of thought / analysis you've already put in (if something doesn't sound right is that because you've only just started thinking about it and there are basic misconceptions, or is it because you've done a really thoughtful analysis and I'm missing a subtlety)?

hartytp, 2021-11-16:
given the above, it's hard for me to say much of any use. Here are a few thoughts off the top of my head, which may or may not be useful...

> The low output power from phaser forces us to chain multiple amplifiers to achieve the desired signal powers. This results in boradband noise from the pre-amplifiers being amplified to appreciable [...]

On one reading of your post, it sounds like you're implying that using multiple gain stages results in more noise. I don't *think* that's what you mean. Is this really two separate points: (a) it's annoying having to have several gain stages (b) having a low output power is bad for phase noise?

On the second of these points: naively I would have thought that the broadband phase noise would be limited by the TRF chip, not a half-decent pre-amp. Is that not the case? And, even if the pre-amp chip does limit the overall system noise, I don't see what we could do about it by modifying phaser. IIRC there aren't other upconverter ICs which would result in lower overall system noise. We could add a pre-amp to phaser, but that wouldn't result in lower noise than an off-card pre-amp.

On the first of these points: what power level do you need, how many amps are you using? Given that these Sinara cards have to server a lot of purposes, we always end up making compromises; what's most convenient for your application will be worse for other applications. when we originally designed phaser we considered a few use-cases. I came out with the impression that adding a pre-amp is annoying for about as many cases as it's helpful (c.f. Urukul where we have a pre-amp + fixed attenuator on the PCB and generally end up running at ~20dB attenuation on the step attenuator to remove the pre-amp gain which we didn't really want in the first place). The other thing we thought about in the design is power consumption: phaser is already pushing how much power it's sensible to burn on a eurocard. Adding high-gain/low-noise pre-amps adds a lot of extra heat, which makes thermal management harder. This was a major motivation in not having a pre-amp (better to let the user sort it out off-PCB where they can pick something appropriate and sort out the heat dissipation)

hartytp, 2021-11-16:
Apologies if I'm answering the wrong question, but putting a few numbers into a point from above...

> On the second of these points: naively I would have thought that the broadband phase noise would be limited by the TRF chip, not a half-decent pre-amp. Is that not the case?

I'm not quite sure which carrier / offset frequency you're interested in and what configuration you're running the upconverter in. But from the [datasheet](https://www.ti.com/lit/ds/symlink/trf372017.pdf?ts=1637021176880&ref_url=https%253A%252F%252Fwww.google.com%252F) the best you're going to get is -160dBc/Hz. If you're getting -10dBm out, that's -170dBm/Hz. So a NF=4dB amp will add 3dB to your broadband noise (i.e. not much, so we'd be milking mice if we tried to improve this!).

Looking at the plots in the datasheet, it's unlikely we're actually hitting the -160dBc/Hz floor (see below for a typical noise plot from the data sheet) in which case, the amp probably doesn't contribute at an easily measurable level.
 
[image]

pathfinder49, 2021-11-16:
> 
> 
> @pathfinder49 I'm a bit confused by your post...(a) because unless I'm missing something it's only tangentially related to the above discussion (b) because there is a lot of context missing about your application/what you're trying to achieve here (c) because it's not clear to me what outcome you want -- is this a question? concrete request for a change to the design? statement about what you would like in an ideal world to inspire future discussion? (d) because I can't tell from the above what level of thought / analysis you've already put in (if something doesn't sound right is that because you've only just started thinking about it and there are basic misconceptions, or is it because you've done a really thoughtful analysis and I'm missing a subtlety)?

Apologies, I should have given more context. The sub -10 dBm output power of Phaser-Upconverter is quite low compared to powers many lab devices such as synths can do. My understanding of the applications for Phaser suggests typical desitred load powers are around +30 to +40 dBm, requiring between 40 and 60 dB gain when starting from a sub -10 dBm signal. Many amplifiers have gains of less than 50 dB, nessecitating multi-stage amplification of the signal. However, each amplifier in the chain will have intrinsic broadband noise reducing the signal to noise below Phaser intrinsic signal to noise. This problem gets worse the more gain you need. 

In my specific case, I am amplifying phaser output to +35 dBm peak signal power using two chained amplifiers. (One amplifier doesn't have enough gain and amplifier choices with sufficent linearity are limited.) When I do not place microwave cavity filters after the amplifiers I measure broadband micowave leakage exciting spectator transitions. In our case this is dominated by pre-amp noise being amplified by the final high-power amp). This is not a killer for us as my usecase allows the use of narrow cavity filters which can selectivly block the broadband noise without incurring too much signal loss.

> (a) it's annoying having to have several gain stages

Pretty much. Especially as all the gain needed to get a suitable output power seems to generate a fair amount of broadband noise. However, I'm yet to check this against theoretical limitations of the amps.

> (c) because it's not clear to me what outcome you want -- is this a question? concrete request for a change to the design? statement about what you would like in an ideal world to inspire future discussion?

This was intended as the latter. The design as it stands. is usable. However, the broadband noise from the required amplifiers isn't inconsequential, so reducing the needed gain would be nice.

> I don't see what we could do about it by modifying phaser

Probably not much with the current choice of components (though the 6 dB loss in the reconstruction filter is unfortunate). Naively I'd think that increasing the DAC singal level into the upconverter would increase the signal to noise ratio.

hartytp, 2021-11-16:
> This was intended as the latter. The design as it stands is usable. However, the broadband noise from the required amplifiers isn't inconsequential, so reducing the needed gain would be nice.

Clarifying language here: the way I read your post, it comes across as though you're saying that the final noise is a function of the number of amplifiers you need. It (almost certainly) is not. All that matters is the lowest level the signal gets to.

> Pretty much. Especially as all the gain needed to get a suitable output power seems to generate a fair amount of broadband noise. However, I'm yet to check this against theoretical limitations of the amps.

This is the really important part. If I'm reading you right, you're making the assumption that (a) the noise in your system is dominated by your pre-amps (b) there is something we could easily do to fix it...but you haven't actually taken any data to verify that or looked into the expected performance of the components to see if it's really true. Is that correct? If so, I'd suggest that we don't have the data to conduct a meaningful discussion.

FWIW though, pre-amps usually come within a few dB of the thermal noise floor, which is pretty low compared with the DAC on phaser. I can't remember off the top of my head what the DAC noise floor is, but I'd guess it's something like -150dBc (maybe worse if you're using the DAC PLL). That's 24dB above the thermal noise floor, so unlikely to be limited by pre-amp noise even at -10dBm output power. Similarly for the upconverter.

If we don't understand what limits we can't find an appropriate solution.

As a piece of context though, Phaser was explicitly not designed as an ultra-low noise board. If we'd wanted to do that, we might well have stuck a low noise PLL on, instead of using the DAC PLL.

> Probably not much with the current choice of components (though the 6 dB loss in the reconstruction filter is unfortunate). Naively I'd think that increasing the DAC singal level into the upconverter would increase the signal to noise ratio (when dominated by amplifier boradband nosie)

Yeah, the reconstruction filter isn't that well optimized and one could probably squeeze out another dB with some work but it feels like milking mice (does a dB or two make that much difference to you?). And, again, I don't think the signal level is actually what dominates the noise so it's a red herring anyway.

hartytp, 2021-11-16:
> Naively I'd think that increasing the DAC singal level into the upconverter would increase the signal to noise ratio (when dominated by amplifier boradband nosie)

Even if the signal level out of the upconverter were the issue, what is your proposal? Put a pair of pre-amps between the DAC channels and the upconverter? What does that do to the heat budget for the design? What do amplifier mismatches do to the IQ mixer performance? What is the max signal level the mixer can take -- can we really boost the signal enough to make a meaningful difference (this would be a lot of faff for a dB or two) given the IC's max input level? I just don't see any concrete workable suggestions here.

hartytp, 2021-11-16:
>Apologies, I should have given more context. The sub -10 dBm output power of Phaser-Upconverter is quite low compared to powers many lab devices such as synths can do.

True, but not a useful data point (it's orders of magnitude smaller and cheaper,, takes a lot less power, has different thermal management, etc).

> My understanding of the applications for Phaser suggests that typical load powers are around +30 to +40 dBm. This requires between 40 and 60 dB gain when starting from a sub -10 dBm signal. Many amplifiers have gains of less than 50 dB, nessecitating multi-stage amplification of the signal. However, each amplifier in the chain will have intrinsic broadband noise reducing the signal to noise below Phaser intrinsic signal to noise. This problem gets worse the more gain you need.

As noted above, I think this statement is false as I read it. The first amplifier is the only one which modifies the S/N ratio. This doesn't get worse with the amount of gain needed.

> In my specific case, I am amplifying phaser output to +35 dBm peak signal power using two chained amplifiers. (One amplifier doesn't have enough gain and amplifier choices with sufficent linearity [...]

We should acknowledge that what you're doing is a very niche/specialist use-case. 10W at `eta~0` is weird, which is why you need a funky cavity filter! We had exactly the same situation using a designed based off of AD9910s in the original microwave trap and those chips are about as low noise as it gets for digital signal generators. Nothing you've said here makes me think that phaser doesn't cover the more standard use cases it was primarily designed for well.

Also, having to have a pre-amp or two to get to nearly 10W doesn't seem surprising or problematic to me. It's a low of power! If gain were free we probably would have boosted the phaser output, but low noise pre-amps are thirsty and we don't have a huge amount of thermal headroom on this design.

pathfinder49, 2021-11-16:
> If I'm reading you right, you're making the assumption that (a) the noise in your system is dominated by your pre-amps

This is not an assumption. To set the total gain we have a 6 dB attenuator in the amplifier chain. Moving this attenuator from before the pre-amp to after the pre-amp reduces the broadband noise by 6 dB. What I have not done is check that the observed spectator excitation rates match up with the pre-amp noise-figure. None the less, the noise does come from the pre-amp. Admittably this noise is likely much less of an issue when using phaser for a laser based gate. For me, it is likely to introduce gate errors on the 1e-4 level. Though at that point we'll probably use a more carfeully selected pre-amp and/or more cavity filtering. 

> Yeah, the reconstruction filter isn't that well optimized and one could probably squeeze out another dB with some work but it feels like milking mice (does a dB or two make that much difference to you?).

Another 5 dB would have made this a non-issue in my current configuration. Swapping the components in the reconstruction filter to be less lossy in the pass-band seems like a fairly straightforward win.  Changing the substrate might also make a difference, though I haven't done the maths, FR4 tends to be lossy for GHz signals. However, as you say, it is a specific application and there are also other ways of tackling the issue. I just wanted to bring up that the sub -10 dBm output powers of phaser aren't ideal. And any signal power that can be gained translates to reduces amplifier noise through the use of attenuators. 

> The first amplifier is the only one which modifies the S/N ratio. This doesn't get worse with the amount of gain needed.

I was under the impression amplifier noise figure tends to get worse the larger the amplifier gain. Though you may know more. 

> Even if the signal level out of the upconverter were the issue, what is your proposal? Put a pair of pre-amps between the DAC channels and the upconverter? What does that do to the heat budget for the design?

As you are much more familiar with the constraints of the phaser design, I primarily wanted to make you aware of some inconveniences of the low output power. This is certainly not a a reason to redesign phaser. I agree, RF amplifiers on Phaser are probably not a good solution. However, if a redesign of Phaser were to happen, it would be nice to consider ways of losing less output power or using a DAC with higher signal level (provided the upconverter can take it).

hartytp, 2021-11-16:
> This is not an assumption. To set the total gain we have a 6 dB attenuator in the amplifier chain. Moving this attenuator from before the pre-amp to after the pre-amp reduces the broadband noise by [...]

That's useful additional data. Taking a step back, I'm not saying that you're wrong here, so much as that there is lots of information missing that makes it hard to give you a useful response.

I still don't understand what you measured or how you measured it (is this an indirect measurement on the ion, or something on a phase noise meter?), exactly what your setup is and how the measurements relate to expectations based on the data sheets from various parts. Without that level of detail / analysis I don't think there is much we can usefully discuss here.

hartytp, 2021-11-16:
> Another 5 dB would have made this a non-issue in my current configuration. Swapping the components in the reconstruction filter to be less lossy in the pass-band seems like a fairly straightforward win

If you think so, give it a go! Propose a design for a reconstruction filter that does the job you're after and give us a simulation.

> Changing the substrate might also make a difference, though I haven't done the maths, FR4 tends to be lossy for GHz signals.

My gut is that there isn't a significant win to be had here and that the cost/benefit analysis of going to a fancy substrate would be horrible. I might be wrong, but someone needs to do some thinking. Apologies if I seem a bit non-receptive to ideas here, but throwing suggestions out with minimal thought isn't productive. The devil is always in the detail and someone needs to do the work. So far I haven't seen convincing data to suggest that the issues you're having really stem from phaser or that there is a practical solution to make them better.

hartytp, 2021-11-16:
> I was under the impression amplifier noise figure tends to get worse the larger the amplifier gain. Though you may know more.

In any well designed low noise system all the noise comes from the first stage. If subsequent stages add noise there is something up with the system design. Think of it this way: noise powers add. Let's say we take a system of two amplifiers with gains `G1` and `G2` and noise powers (as measured at the amplifier input) `N1` and `N2`. Ignoring noise from the source, the output noise is `G1*G2*N1 + G2*N2`. If we refer this to the input of the first amp, the noise is `N1 + N2/G1`. i.e. the effective NF of the second amp is divided by the gain of the first amp. By the time you've gone through 20dB of gain, it becomes hard to build an amp with such a bad NF that it has a significant impact on the overall signal noise.

hartytp, 2021-11-16:
> As you are much more familiar with the constraints of the phaser design, I primarily wanted to make you aware of some inconveniences of the low output power. This is certainly not a a reason to [...]

We did actually think about the output power and we had a few discussions about it. The short answer is that there isn't a goof option for a DAC with significantly higher output power (if you're struggling for a couple of dB you may be able to win by looking at the DAC biasing...there might be some room to play with, I can't remember).

hartytp, 2021-11-16:
> Apologies if I seem a bit non-receptive to ideas here, but throwing suggestions out with minimal thought isn't productive. The devil is always in the detail and someone needs to do the work. So far [...]

Putting this a different way: communication (and the framing around communication) is hugely important to these open source projects. It's easy for a comment that is intended in the spirit of "I haven't thought about this for more than 5 min, but here is a data point that may or may not be useful" being interpreted as "this is something I've thought through really well and we should definitely do it". This ends up in designs becoming bloated and unusable because of a bunch of ill-thought through additions. So I'm generally really keen to drill into any posts to understand what the issue really are and how much thought has gone into things. and, also, to make sure we are very clear to signpost when no further action can be taken without more thought / data.

From what you've said, it sounds like this is more in the former camp; there may be something here, but more work and explanation is needed before this could lead into a potential design change. If you want to give this more thought and come back with some more detailed analysis / concrete proposals then great (although, let's do that in a new thread). Otherwise, let's leave this here