sinara-hw/Phaser #132: LED map

Jacky2k, 2021-04-14:
I am looking for some way to figure out which LED is which. There are no labels on the PCB and no extra documentation (that I could find). The Altium files are not an option as I don't have access to an Altium license.

jordens, 2021-04-14:
The PDF of layouts and schematics are here. And Altium365 works just fine.
As mentioned before it would be great and relatively easy if someone sets up some github pages to generates the html pages for the viewer iframes for all releases of all the sinara-hw projects automatically and even annotate the releases with those links.

dtcallcock, 2021-04-14:
At least in the US, even full Altium licenses are free for students.

Jacky2k, 2021-04-14:
> The PDF of layouts and schematics are here.

Where are the PDF for the layouts? There are only two PDFs I could find, which contain the two schematics. Additionally there are gerber files, which do not contain the labels for the LEDs. So I am unable to link both together.

> And Altium365 works just fine.

Yes, and as far as I could figure out, it costs 3495€ per year.

> At least in the US, even full Altium licenses are free for students.

I am not a student. Also, I think installing a full Altium just to find out where the LEDs are is a bit too much.

I would prefer the LEDs to be labeled on the board, which are probably the most important labels besides the MMCX connectors. Or at least a population map, showing the position of those important components.

gkasprow, 2021-04-14:
Right, I forgot to annotate LEDs on the front panel.
[image]

gkasprow, 2021-04-14:
It would be better to assign more specific functions to LED0....LED5

gkasprow, 2021-04-14:
[image]
[image]

Jacky2k, 2021-04-15:
Well, I am still not 100% happy ;)
It is still unclear what A and B of those are and the LEDs on the board are also unlabeled.

jordens, 2021-04-15:
@Jacky2k (Fabian, right? It would be good to let everyone know who you are. Otherwise it's a lot of guesswork for those trying to help you).
Altium 365 Viewer (the one you'll find quickly googling for "altium 365") is indeed free. You don't need to install or buy Altium.
If you look closely at the schematics you already found, the LEDs are clearly marked TOP and BOT. If that's unclear: BOT is the one close to the PCB.

[image]

Jacky2k, 2021-04-15:
> Fabian, right? It would be good to let everyone know who you are. Otherwise it's a lot of guesswork for those trying to help you

Yes. I would love to change my github user name, but sadly, the last time I checked, it was not possible without breaking every link all over github to my account.

> If you look closely at the schematics you already found, the LEDs are clearly marked TOP and BOT.

Ohh, I missed that. Thanks a lot! But the LEDs on the PCB are still unlabeled.
Actually, I don't understand why this is such a big discussion. I think this is a bug in the design, which did cost me quite some time to work around, so I reported it. If you think this is not a bug or don't want to solve it for whatever reason, say that and close the report. No big discussion required.

jordens, 2021-04-15:
None of this is a big discussion. Yes, the issue is valid (thanks for filing it) and there should be labels on the PCB and on the panels. As you can see from Greg's screenshots, he thankfully already fixed it for the next release. You'll understand that new PDFs will only be posted when preparing for a new release.
In addition to addressing the LED labeling issue from multiple angles, there was a strong need to address and correct your secondary lamentation about having no access to the design files, about the Altium cost and availability, about the availability of documentation, and about the content of the schematics.

jordens, 2021-04-15:
If you want, you can always set a full name independent of you user handle.

closed 2021-04-26 by gkasprow