sinara-hw/Phaser #133: SI with high VSWR

pathfinder49, 2021-04-15:
# Summary
When using Phaser-Upconverter (haven't tested baseband) with a high VSWR load, there seem to be SI issues.

## Details
If I produce maximum power RF at ~3 GHz with a slightly loose SMA connection at the RF output, there are sporadic errors when initialising Phaser. The error message suggests there may be a signal-integrity issue. I only seen this issue occur with an non-terminated or poorly tightened output connection.

The error message is: `ValueError: large number of frame CRC errors`

I struggle to reproduce the error reliably. It tends to occur sporadically in the situation described above.

I don't see this as a particularly large issue. However, it may be worth checking if there is an easy fix.

jordens, 2021-04-15:
The CRC errors are on the Kasli-Phaser link. The counter counts up from the time Phaser locks to the data clock the first time and is never reset. The error will be permanent until a Phaser reset (through loss of lock or through phaser reboot). 
For the RF side to cause SI issues on the data link, I would expect a significant level of ground immission or ESD. That would be worrying.
If we determine that CRC errors are not an issue or not at this level, then we could change the init code to just count the new crc errors between start of init and end of init.

gkasprow, 2021-04-26:
If the RF path is closed by EEM link, that's not a surprise that you get CRC errors :)

jordens, 2022-09-15:
Not reliably reproducible, attributed to loose connection and RF immission.
No bandwidth and interest by author to provide input. Closing.

closed 2022-09-15 by jordens