sinara-hw/Phaser #135: IC10 unavaliability

pkozakiewicz, 2021-04-26:
Please find an alternative for SC18IS602B bridge in the new release. This component became very hard to purchase. Maybe double footprint should be considered.

gkasprow, 2021-04-26:
I think we can use SC18IS602IPW 
SC18IS602IPW does not support SS2. This pin should be used as GPIO2 only. We don't care about SS2 and use it as GPIO.

pkozakiewicz, 2021-04-26:
SC18IS602IPW is an obsolete.

gkasprow, 2021-04-26:
the same with SC18IS603 which could be potentially used

gkasprow, 2021-04-26:
this is not a critical part - it's used only during remote FPGA programming. The lead time is 52 weeks.
What about this source?
https://www.win-source.net/interface-controllers-sc18is602bipw112.html

pkozakiewicz, 2021-04-27:
We will check, but for the future I would prefer to have something available in the EMEA market. It can take weeks for packages from HK to arrive.

jordens, 2021-04-27:
IIRC we use that chip on Banker, Humpback, Fastino and Phaser. It would be unfortunate if we have to redesign the hardware and software.

gkasprow, 2021-04-27:
It is still produced, but recently electronics production is tough because many components have availability issues. Even when we redesign it, there is no guarantee that after a month or so the issue won't reappear. The only remedy is to keep the stock of hard to get components.

gkasprow, 2021-04-27:
It is normally available in Mouser/Farnell and other stores. As I said, recently there are shortages of very components.

closed 2022-09-09 by gkasprow