sinara-hw/Phaser #136: Expected LO level

Jacky2k, 2021-04-27:
Hi,
I just noticed that there is a note in the schematic: "Expected LO level is -10dBm" for the TRF. The datasheet does not say anything about the expected LO value. However, the LO output level is 3 dBm and I would expect the input to expect a similar value. This assumption is also confirmed by TI: https://e2e.ti.com/support/rf-microwave/f/rf-microwave-forum/984697/trf372017-ext_vco-maximum-input-level-of-trf372017
So, is there any reason to use -10 dBm instead that I missed?

gkasprow, 2021-04-27:
I think I calculated the power from an input voltage

Jacky2k, 2021-04-27:
Where did you got that input voltage from? I cannot find anything in the datasheet.
Interestingly, when I reduce the LO power below -5 dBm the sideband suppression of the mixer gets significantly worse. At -10 dBm I have less than 30 dB suppression left. LO is at 300 MHz. So I guess -10 dBm is way too low. On the other hand, increasing the level beyond +2 dBm also decreases the sideband suppression. So I guess a value between -4 and +2 dBm is optimal.

gkasprow, 2021-04-27:
I took it from this image :)
[image]

Jacky2k, 2021-04-27:
I guess you are talking about the Po=-10dBm value?
My guess would be, that this is the output power after the mixer. Especially, as Po is also given in figure 27, which is using the internal VCO.

gkasprow, 2021-04-27:
Yes, I wasn't sure if it is VCO Po or IC output

Jacky2k, 2021-04-27:
OK, then I would recommend updating the schematic note.
One may also argue to remove the attenuator in front of the second LO input.

gkasprow, 2021-04-27:
It makes sense.

gkasprow, 2022-09-09:
The attenuator is removed by default. I also removed the schematic note.

closed 2026-09-17 by kaolpr