sinara-hw/Phaser #137: Position of LO termination resistors

Jacky2k, 2021-04-27:
Hi everyone,
I noticed that the termination resistors, which are involved in the LO signals of the TRFs are quite far away from where they should be. This involves the resistors R226A/B, R224A/B, and R225A/B. As far as I currently understand the LO signal may be as high as 4.8 GHz. This results in a wavelength of approximately 10mm and the distance of these resistors to their corresponding input/output is in the same range. This does not seem right.

gkasprow, 2021-04-27:
Are you sure you are looking at the newest revision? It was fixed in v1.1

gkasprow, 2021-04-27:
Ah, in rev1.1 I fixed the DAC termination resistors. 
IN the case of TRF, look how it's done on devkit.