sinara-hw/Phaser #138: TRF LO output level varies when loading second output

Jacky2k, 2021-04-27:
Hi everyone,
I noticed a strange behaviour of the LO output levels of the TRF: When I connect a 50 Ohm load to the LO OUT N of the first TRF (J14B), the amplitude of the internal LO connection (connected to LO OUT P) varies significantly (something like 15 dB!). Interestingly the amplitude of P is higher, when the N output is loaded.
I am not 100% sure where this is coming from. It may be, that the differential output does not like being loaded differently, I could not find any information in the datasheet about this.
I have to mention, that I replaced the 10 dB attenuator with a 0 Ohm resistor, according to the discussion in https://github.com/sinara-hw/Phaser/issues/136. I will change this back shortly, and see if that makes any difference.
Any idea why the board is behaving like this?

gkasprow, 2021-04-27:
are you using AC or DC connection? I mean before or after the coupling cap ?

Jacky2k, 2021-04-27:
I measured at the input of the the 10 dB attenuator (or my 0 Ohm resistor).
Putting the attenuator back in place reduces the effect to ~6 dB, but that is still a problem.
I noticed, that TI put a 100 Ohm resistor between N and P in the eval board, that is the only difference I can see.

Jacky2k, 2021-04-27:
I am an idiot and I am sorry.
I had a wrong power value set in my RF generator all the time, the LO outputs work as expected.
Once everything is setup properly I measure roughly -2 dBm under all circumstances.

closed 2021-04-27 by Jacky2k