sinara-hw/Phaser #141: Phaser 1.0: Reference clock leakage into RF outputs

pathfinder49, 2021-04-30:
There is appreciable leakage of reference clock harmonics into the RF outputs. There leakage is found wheth the mmcx or SMA clock are used and in both the Baseband and Upconverter variants.

# Overview
Clock leakage at 125 MHz is quite low at -100 dB. However, the leakage increases at higher frequencies. At the 3.125 GHz harmonic, the attenuation of the external clock to the outputs is only -40 dB. 

Clocking Phaser with a ~2.3 dBm square wave, results in approximately -75 dBm output spurs at 3.125 GHz. Using The Kasli-clock via MMCX results in a -85 dBm spur in channel 1. For context the upconverter maximum output power in much of its f_lo +- 500 MHz range is around -20 to -30 dBm. These clock spurs are only 45 dB below this signal power and are close to the upconverter signal frequencies. 

This is not particularly concerning for the baseband variant as these high frequency spurs can be filtered out. 

# Details
The 3.125 GHz output-spur amplitudes on phaser-upconverter depend on whether the MMCX or front-panel inputs are used.

If the front panel input is used, the spur amplitudes do not depend on whether phaser is power. These can be observed by simply connecting a phaser board to a clock input and a spectrum analyser to the chosen output. The coupling to ch0 and ch1 is -47 and -37 dB respectively. These do not change appreciably when phaser is powered. 

The MMCX-input leakage on an un-powered phaser is much less than that of the SMA input (roughly -70 dB). When powering the board (and selecting the MMCX clock) this increases to roughly -50 dB for both output channels. 

# Notes
- The front-pannel clock is associated with the largest coupling by roughly 20 dB. 
- To me, the MMCX behaviour suggests that the clock distribution on phaser is responsible for coupling around -50 dB.
- The issue can be worked around by passing the clock through a low-pass filter before feedig it to phaser.

gkasprow, 2021-04-30:
This can be caused by additional RC impedances between the SMA  ground and board GND.
Do we need these "floating connector ground" features? It may cause more harm than good.
You can try to connect the SA between board GND, and the clock SMA ground?
[image]

hartytp, 2021-04-30:
> Do we need these "floating connector ground" features? It may cause more harm than good.

If they're causing issues, I'd be very happy to scrap them. We started doing this a long time ago and should probably re-evaluate whether they're really the right thing to do

pathfinder49, 2021-04-30:
Another thought: When looking through the schematics, I noticed the front panel clock changes ground reference layer without any nearby ground vias. IIRC for EMI differential pairs couple quite similarly to individual traces (an example is next to TR2 in the picture above). This may cause return currents to spread out.

I don't have access to Altium right now, so could provide images later.

pathfinder49, 2021-04-30:
> You can try to connect the SA between board GND, and the clock SMA ground?

I'd rather keep re-working to a minimum. Replacements seem to be hard to come by :(

gkasprow, 2021-04-30:
I mean to connect SA ( Spectrum Analyzer) and measure the SMA ground bouncing

gkasprow, 2021-05-05:
I removed the SMA "AC isolation" components. That should help.

gkasprow, 2021-05-05:
@pathfinder49 can you make to clock input floating? To do so, remove C37, R14, and R81. That would reduce the ground loop and possibly lower the leakage.

gkasprow, 2021-05-05:
Actually, you need to remove only R81, other components are DNP already

jordens, 2022-09-15:
No bandwidth and interest by author to provide input. Closing.

closed 2022-09-15 by jordens