sinara-hw/Phaser #148: DAC and up-converter mismatched, no support for DC

jbqubit, 2022-02-10:
Our choice of DAC ([DAC34H84](https://www.ti.com/product/DAC34H84)) and PLL-Mixer ([TRF372017](https://www.ti.com/product/TRF372017)) preclude passing DC to the TRF. AFAICT TI has specific recommendations for DAC-PLL/mixer pairings but we didn't follow it. I don't recall the design choices that led to this pairing but it appears there is no resolution possible with the current chip pair. 

# current filter network
Phaser v1.1 coupling between DAC and TRS looks like this. 
[image]

This matches (approximately) the recommendation of the specification sheet for the TRF. In this approach the TRF chips provides it's own bias (1.7 V) to the I and Q inputs. In this mode the TRF also has DACs for trimming needed to achieve carrier feedthrough cancellation. However, this approach precludes IQ modulation that depends on DC from the DAC. 

[image]
[image]

# DAC can't supply 1.7V
TI provides an example filter network in the case that the DC-TRF coupling is DC-pass. 

[image]

However, we can't use this approach because the maximum output of our DAC is -0.5 V and 0.6 V (for the differential outputs, spec sheet 6.5 Output compliance range). 

# pull-up resistor option
A TI [application note](https://www.ti.com/lit/ug/slua647a/slua647a.pdf?ts=1643313111661&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FDAC34H84
) discusses building a filter network with pull-up resistors (p 7). This would attenuate the DC component of the DAC by at least 0.5. 

# one motivation: PDH ESB locking
One application of Phaser is to generate phase modulation suitable for PDH electronic sideband locking (ESB) [0]. 
[image]
[image]

A way to do this with Phaser is phase modulated I and Q. However, this has a DC component that is blocked by the current filter network.

[image]

# Ref 
[0] Thorpe, J. I., Numata, K. & Livas, J. [Laser frequency stabilization and control through offset sideband locking to optical cavities](https://www.osapublishing.org/oe/abstract.cfm?uri=oe-16-20-15980). Opt. Express 16, 15980 (2008).

jbqubit, 2022-02-10:
How flat is the response of this approach for frequencies below cutoff of the 1 uF cap?

gkasprow, 2022-02-11:
Should be flat up to dc. May need LC tuning. I can simulate it if you consider this solution.

jbqubit, 2022-02-11:
> May need LC tuning.

In an IQ context the phase response is important. 

@restelli may have comments on the current bias approach you've suggested.

hartytp, 2022-02-14:
@jbqubit apologies if this comes out a bit negative. It's been my repeated experience with these Sinara boards that one needs to constantly fight scope creep. These boards aren't going to cover every use and that's okay. If you want a more major design change, feel free to fork the project or look at one of the other sinara designs like Sayma which has a wider remit.

closed 2022-02-14 by hartytp

jbqubit, 2022-02-15:
What is the -3dB corner for the current coupling between DAC and TRS? I'll add this to the wiki documentation for v1.0 and v1.1.