sinara-hw/Phaser #153: Question: Remove upconverter?

pathfinder49, 2022-09-14:
Is the on-board upconverter actually useful for people? In my experiment the baseband version with a passive external IQ mixer performs much better. Therefore, I wonder if it is worth keeping around.

The Phaser upconverter variant has fairly low output power (#130) and some signal to noise limitations (#141 & ~20 dB signal to LO feed-through ratio). 

Is there a project benefit to removing the up-converter? As I understand it the up-converter power consumption is part of the reason for not using an external DAC PLL (#2) or greater output signal.

gkasprow, 2022-09-14:
That's we we have the Baseband assembly variant without the upconverter.
What type of the upconverter do you use? Maybe there is something wrong with the design, for example noisy power supply lines...
Moreover in previous revision there was a bug in the power supply voltage which could affect the output power level.
Maybe we should integrate the passive upconverter instead..

hartytp, 2022-09-14:
> Is there a project benefit to removing the up-converter? As I understand it the up-converter power consumption is part of the reason for not using an external DAC PLL (https://github.com/sinara-hw/Phaser/issues/2) or greater output signal.

I'm afraid I don't follow this discussion. How would we get a larger output signal? You mean with a pre-amp? How does that improve the signal to noise?

What kind of PLL do you have in mind and what problem would it solve?

jordens, 2022-09-14:
Bad idea.
It is very useful to us and others.
There is a variant (there is no point in removing the upconverter since it doesn't impact you if you don't want it).
Even if your analysis is correct removing the upconverter would not help you in any way.
Let's sort out your measurements before drawing conclusions.

pathfinder49, 2022-09-14:
> I'm afraid I don't follow this discussion. How would we get a larger output signal?

A DAC with higher output voltage might help with some signal to noise aspects, but it would be a fairly invasive change which is probably not worth it. To me the point of opening the issue is to get a better impression of what people are interested in and if the upconverter limitations were too severe for other users. 

> Bad idea. It is very useful to us and others. There is a variant (there is no point in removing the upconverter since it doesn't impact you if you don't want it). Even if your analysis is correct removing the upconverter would not help you in any way. 

Good to hear it's being used 😄 In that case it is of cause best to keep it. The only argument against the TRF I'm aware of is keeping space for it in the power budget, which isn't a huge issue unless we want to use it for something else.

> Let's sort out your measurements before drawing conclusions.

I currently have no capacity for further phaser characterisation. The observations are experimental issues I encountered on the way to attaining satisfactory performance in my setup, YMMV.

> What kind of PLL do you have in mind and what problem would it solve?

Not something I need, but something that seemed to come up in discussions and seemed to be discarded because of the power budget. In my applications I'm not limited by the DAC phase noise.

From these responses it seems like there is little appetite increased power DACs or improved phase noise at the expense of the upconverter.

closed 2022-09-14 by pathfinder49

pathfinder49, 2022-09-14:
> What type of the upconverter do you use? Maybe there is something wrong with the design, for example noisy power supply lines... Moreover in previous revision there was a bug in the power supply [...]

I use a  table-top synth as the 3 GHz LO and the [Eclipse IQ2040MP4](https://www.eclipsemdi.com/product/2-0-4-0-ghz-iq-mixer-iq2040mp4/ 
) passive mixer, admittedly a much more expensive solution and not a fair comparison.

hartytp, 2022-09-15:
> A DAC with higher output voltage might help with some signal to noise aspects, but it would be a fairly invasive change which is probably not worth it. 

IIRC the DAC on phaser is a standard 20mA FS output stage (same as the AD9910 for example). I'm not aware of any real options for DACs with higher output drive levels.

> From these responses it seems like there is little appetite increased power DACs or improved phase noise at the expense of the upconverter.

The other point I'd generally make here is that there isn't any fundamental noise advantage to be had from using a higher-power DAC. As @jordens pointed out already, the DAC's noise floor is a long way above the thermal limit so a DAC with higher output power but the same noise specs would not outperform the current DAC + a pre-amp.

> Not something I need, but something that seemed to come up in discussions and seemed to be discarded because of the power budget. In my applications I'm not limited by the DAC phase noise.

It came up a few times, but I don't recall seeing detailed analysis that made a convincing case for it actually being helpful.

> Maybe we should integrate the passive upconverter instead..

The upconverter IC we have on phaser already /should/ be excellent. e.g. much lower noise than the DAC. But, also the carrier feedthrough specs are better than I've seen on passive IQ mixers. For example, the feedthrough specs are better than the passive mixer that @pathfinder49 linked to. If we're not coming close to hitting those specs then it sounds like a design issue rather than an issue with the mixer itself.

gkasprow, 2022-09-15:
The upconverter IC we have on phaser already /should/ be excellent. e.g. much lower noise than the DAC. But, also the carrier feedthrough specs are better than I've seen on passive IQ mixers. For example, the feedthrough specs are better than the passive mixer that @pathfinder49 linked to. If we're not coming close to hitting those specs then it sounds like a design issue rather than an issue with the mixer itself.

 I'd like to recreate the issue then to study it and see how to fix

pathfinder49, 2022-09-15:
Judging from my notes when I was working on this the upcoverter:
1. The upconverter LO feed-through is ~-40 dBm (as stated in the data sheet). 
    - Adjusting the current offsets I could get this to ~-60 dBm. So the LO leakage can be quite good with the TRF. Anecdotally when compensated to -60 dBm leakage the compensation drifted. Therefore, ~-50 dBm is a bit more realistic. 
    - The signal level in the up converter variant is <-20 dBm for realistic use cases, so at best the LO feed-through to signal ratio is somewhere between 20 and 40 dB (depending on manual tuning)
2. Using the passive external mixer I found that, in practice, I get -40 dBm LO feedthrough when using the passibe mixer. (This is better than the 40 dB stated in the data sheet).
    - As the baseband output power is appreciably higher, the available signal power is ~-8 dBm (more like -14 dBm if you want two tones). Therefore, the signal to LO leakage ratio ratio is ~30 dB. Overall this is comparable to the TRF.
3. Broadband noise from the table-top synth is significantly lower than from the TRF. 
    - In particular the TRF noise floor at >10 MHz detuning is higher than external LOs can provide and creates a braodband noise floor (see https://user-images.githubusercontent.com/14295481/114721023-6c8b6d80-9d30-11eb-8737-0e9b3ce092f6.PNG)
    - Using an external synth at 3GHz the >10 MHz detuned noise can be as low as -160 dBc/Hz (R&S SMA100A). As the signal to leakage ratio is comparable between the TRF and external upconverter. This should give a ~20 dB drop in LO-leakage broadband noise when compared to the signal strength. However, I did not characterise this background drop in isolation.

Using an external upconverter with a cleaner LO also yielded noise advantages at smaller detuning. Besides the coherence time issues caused by broadband noise, switching to an external upconverter. Resulted in a ~50x error reduction in single qubit randomised benchmarking (qubit logic operations) within my experiment. However, as this isn't the focus of my work I haven't dug into the detailed causes of this (close in noise vs broadband noise issues). 

> Moreover in previous revision there was a bug in the power supply voltage which could affect the output power level.

The Phaser upconverter I was using was effected by this and had the white wire fix to the power supply. I don't think the upconverter was damaged, as it generally performs as the spec sheet would have me believe and it was [within the maximum specified limits of the pins](https://github.com/sinara-hw/Phaser/issues/130#issuecomment-816887020).

For context, here are some spectra I have on hand with Phaser baseband & external upconversion vs Phaser upconverter.

| Phaser baseband and SMA 100A with passive upconverter | Phaser upconverter |
| ---- | ----|
| [image] | [image] |