sinara-hw/Phaser #157: SMA connectors moved too far outside

jmoskwa, 2023-12-12:
Move all the SMAs towards inside of PCB 2,3mm from current position. Reason - no room for isolating sleeves (marked with small arrows) and collision between pins and edge of the panel.
[image]
[image]

jordens, 2023-12-12:
When changing this, please seize the chance to check that there's a coherent and good approach to this on all modules (Kasli, Kasli-SoC, Stabilizer, Phaser, Urukul, Mirny, Pounder, DIO-SMA, IDC-SMA, and others) w.r.t. part-number, isolation, hole size, RFI lip isolation, nut/washer usage, spacing from panel etc.

jmoskwa, 2023-12-12:
Yes, I will.

pkozakiewicz, 2023-12-15:
@jmoskwa In TS we are already working on introducing consistent panel spacing in all modules, i.e. last Stabilizer:

[image]

The main principle is that signal pin in all SMA connectors (DELOCK 65848) should be moved 1.9mm away from the mounting hole (as shown in the attached picture) so that there is 8.5mm thread outside the front panel left. All of the devices we have released recently comply with this.

jmoskwa, 2023-12-15:
Ok, thank you for this information. I will take it into account.

gkasprow, 2023-12-15:
make sure that SMA connector nut does not touch the panel

jmoskwa, 2023-12-15:
In this project or all with SMAs?

gkasprow, 2023-12-15:
all of them
Some time ago we moved all connectors to the panel to make sure the nut doesn't touch it
Then we switched to DELOCK connectors in all new revisions.

pkozakiewicz, 2023-12-15:
FYI We do not use SMA nuts anymore. The only element between the connector and the panel is fdm printed sleeve mounted from the backside.

gkasprow, 2023-12-15:
I meant such SMA nuts :)
[image]

kaolpr, 2026-09-17:
@gravelpitman Can you please comment on this?

gravelpitman, 2026-09-17:
As @pkozakiewicz mentioned earlier, we have been using standard X axis positioning for a long time, which gives us consistent SMA washer and SMA stickout. This is documented in the guidelines at https://github.com/sinara-hw/meta/wiki/StyleGuide 
IMO this issue may be closed.

closed 2026-09-17 by gravelpitman